m5stack/M5Stack-Camera #3: Error 0x20001

sige1, 2018-12-20:
D (205) spiram: Allocating block of size 32768 bytes
D (215) intr_alloc: Connected src 16 to int 12 (cpu 0)
D (225) nvs: nvs_flash_init_custom partition=nvs start=9 count=6
I (265) camera: Enabling XCLK output
D (265) ledc: LEDC_PWM CHANNEL 0|GPIO 27|Duty 0002|Time 0
I (265) camera: Initializing SSCB
I (265) camera: Resetting camera
I (265) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (295) camera: Searching for camera address
E (305) camera: Camera probe failed with error 0x20001
E (315) camera: Camera Init Failed

I use it.
M5CAMERA001
https://www.switch-science.com/catalog/5207/

I think pin assignment is wrong
Please tell me the correct pin assignment

BR.

sakabin, 2018-12-24:
hello, it need to change 
```c
#define CAM_PIN_SIOD    22
#define CAM_PIN_VSYNC   25

uart_set_pin(UART_NUM_1, 13, 4, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
```

closed 2019-01-02 by sakabin

sige1, 2019-01-05:
Thank you for your reply.

I was able to confirm the fix.

I feel happy!

BR.