m5stack/M5Stack-Camera #6: black screen on start

machinehistories, 2019-03-19:
I am having an issue with the new fisheye camera. I plugged it in and accessed the ap wifi but then when I go to 196.128.4.1 it is just showing a black screen with a small white square in the middle. I downloaded the firmware and reloaded it but it does the same thing. Can anybody tell me what might be the issue.

sakabin, 2019-03-19:
Does it work when you get the camera at the beginning?

machinehistories, 2019-03-19:
No when I first purchased it the screen was black with small white square.
I have successfully used my camera model A but I haven't been able to get
the model F to work. When I reinstall firmware it looks exactly the same
which is to say it has a black screen and small white square.

sakabin, 2019-03-19:
If this problem has not been resolved, please contact the seller to change
a new one

closed 2019-03-29 by sakabin