m5stack/M5Stack-Camera #7: How to set TXD/RXD/RTS/CTS for UART

Herwey, 2019-03-29:
Hi
Now I am using M5 model A for UART, how to set TXD/RXD/RTS/CTS ?? I set it as below:
#define ECHO_TEST_TXD  (28)
#define ECHO_TEST_RXD  (14)
#define ECHO_TEST_RTS  (12)
#define ECHO_TEST_CTS  (13)

But it not worked very well

sakabin, 2019-03-29:
In uart branch, test it
https://github.com/m5stack/m5stack-cam-psram/blob/uart/main/main.c#L123

Herwey, 2019-03-29:
Thanks ! I used this , but nothing printed in serial....

sakabin, 2019-03-29:
https://github.com/m5stack/m5stack-cam-psram/blob/uart/main/main.c#L132
Serial print in tx=13, rx=12, not in usb-c bus

Herwey, 2019-03-29:
I see, I used "make monitor".  When I open a serial tool, I set the "baud_rate = 921600", it was wrong, it's not as below 
        uart_write_bytes(UART_NUM_1, (char *)tx_buffer, 7);
        uart_write_bytes(UART_NUM_1, (char *)fb->buf, fb->len);
When I set "baud_rate = 115200", I can see the print info, but not serial.Right?

Herwey, 2019-03-30:
I set the pin as below:
uart_set_pin(UART_NUM_0, 13, 12, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE);
It can send the data from serial, but it can't read the data from serial.
emmm...