m5stack/M5Stack-Camera #8: Failed to init external RAM!

fah, 2019-03-29:
I bought this camera:
https://www.aliexpress.com/item/M5Stack-Official-ESP32-Camera-Module-Development-Board-OV2640-Camera-Type-C-Grove-Port-3D-Wifi-Antenna/32881414545.html

I've set 
D2, VSYNC, SIOD and camera type to 1 in main.c
```

#define M5_CAM_KIND 1 // 1 --> A model, 2 --> B model
// #define FISH_EYE_CAM  // fish eye need flip image
#define CAM_USE_WIFI

#if M5_CAM_KIND == 1
#define CAM_PIN_SIOD    25
#define CAM_PIN_VSYNC   22
#else
#define CAM_PIN_SIOD    22
#define CAM_PIN_VSYNC   25
#endif

//M5STACK_CAM PIN Map
#define CAM_PIN_RESET   15 //software reset will be performed
#define CAM_PIN_XCLK    27
#define CAM_PIN_SIOC    23

#define CAM_PIN_D7      19
#define CAM_PIN_D6      36
#define CAM_PIN_D5      18
#define CAM_PIN_D4      39
#define CAM_PIN_D3      5
#define CAM_PIN_D2      34
#define CAM_PIN_D1      35
#define CAM_PIN_D0      17

#define CAM_PIN_HREF    26
#define CAM_PIN_PCLK    21

```
I do not know what to enter for modbus as I don't use it I set it to 11,12 and 14 in sdkconfig.
Like so
```
CONFIG_MB_UART_RXD=11
CONFIG_MB_UART_TXD=12
CONFIG_MB_UART_RTS=14
```

I got this error:

```
...
I (639) boot: Loaded app from partition at offset 0x10000
I (639) boot: Disabling RNG early entropy source...
D (639) boot: Mapping segment 0 as IROM
D (643) boot: Mapping segment 4 as DROM
D (647) boot: calling set_cache_and_start_app
D (652) boot: configure drom and irom and start
D (656) boot: start: 0x400813b0
0x400813b0: call_start_cpu0 at /Users/f/esp/esp-idf/components/esp32/cpu_start.c:120

E (660) cpu_start: Failed to init external RAM!
abort() was called at PC 0x4008140d on core 0
0x4008140d: call_start_cpu0 at /Users/f/esp/esp-idf/components/esp32/cpu_start.c:171 (discriminator 1)

Backtrace: 0x40093058:0x3ffe3bd0 0x40093281:0x3ffe3bf0 0x4008140d:0x3ffe3c10 0x40079202:0x3ffe3c30 0x400792f9:0x3ffe3c60 0x40079317:0x3ffe3ca0 0x400796fb:0x3ffe3cc0 0x400807d2:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
0x40093058: invoke_abort at /Users/f/esp/esp-idf/components/esp32/panic.c:680

0x40093281: abort at /Users/f/esp/esp-idf/components/esp32/panic.c:680

0x4008140d: call_start_cpu0 at /Users/f/esp/esp-idf/components/esp32/cpu_start.c:171 (discriminator 1)

```

***What do I miss ?***

```
$ make version
Toolchain path: /Users/f/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
```

sakabin, 2019-03-30:
Your camera not have Psram, try https://github.com/m5stack/m5stack-cam-psram/tree/NoPsram

fah, 2019-03-31:
You are right. I thought the SRAM is the same as PSRAM. Thank you.

closed 2019-03-31 by fah