m5stack/M5Stack-Camera #11: Camera not detected after esp_restart

falkoschindler, 2019-04-10:
I'm trying to restart the ESP programmatically using `esp_restart()`. But after doing so, the camera can't be initialised. These are the last log messages after rebooting:

```
I (628) cpu_start: App cpu up.
I (1135) spiram: SPI SRAM memory test OK
I (1135) heap_init: Initializing. RAM available for dynamic allocation:
I (1135) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1141) heap_init: At 3FFBB2A0 len 00024D60 (147 KiB): DRAM
I (1148) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1154) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1161) heap_init: At 40097998 len 00008668 (33 KiB): IRAM
I (1167) cpu_start: Pro cpu start user code
I (1172) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (186) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (187) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (237) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (257) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (297) camera: Camera probe failed with error 0x20001
E (297) camera: Camera Init Failed
```

The issue can be reproduced inserting the command `esp_restart();` within an http handler, e.g. in `imu_handler` on the origin/mpu6050 branch and requesting the imu endpoint.

Is there anything I can do to restart the ESP on demand?

sakabin, 2019-04-10:
Try add ```vTaskDelay(100/portTICK_PERIOD_MS)``` , it will be ok :)
[image]

falkoschindler, 2019-04-10:
Thanks for your quick reply!

Unfortunately, `vTaskDelay(100/portTICK_PERIOD_MS)` or even `vTaskDelay(1000/portTICK_PERIOD_MS)` does not help.

sakabin, 2019-04-10:
I found the problem, Try test it https://github.com/sakabin/m5stack-cam-psram/tree/mpu6050
These are all needed
[image]

falkoschindler, 2019-04-10:
Awesome! That solved my problem.

I could even get it wo work with these lines only:

```
gpio_reset_pin(GPIO_NUM_23);
gpio_reset_pin(GPIO_NUM_22);
esp_restart();
```

This avoids the need for `imuTaskExit()`, which is a bit inconvenient to implement.

closed 2019-04-10 by sakabin