m5stack/M5Stack-Camera #12: How to read the JPEG data directly??

Herwey, 2019-04-11:
fb = esp_camera_fb_get();   //获得buffer指针
        if (!fb) {
            ESP_LOGE(TAG, "Camera capture failed");
            res = ESP_FAIL;
        } else {
            if(fb->format != PIXFORMAT_JPEG){
                bool jpeg_converted = frame2jpg(fb, 80, &_jpg_buf, &_jpg_buf_len);  // 获取图像buffer
                if(!jpeg_converted){
                    ESP_LOGE(TAG, "JPEG compression failed");
                    esp_camera_fb_return(fb);
                    res = ESP_FAIL;
                }
            } else {
                 _jpg_buf_len = fb->len;
                 _jpg_buf = fb->buf;
            }
Now, I want to read the _jpg_buf directly(RGB), how to do this ? Can I ?  I want to transfer the JPEG by spi.

sakabin, 2019-04-11:
sorry，i don`t konw what you want to, rgb or jpeg data?  if jpeg,  just need transfer fb buffer by spi, like this https://github.com/m5stack/m5stack-cam-psram/blob/uart/main/main.c#L108, if rgb, I don't recommend generating rgb data via spi

Herwey, 2019-04-11:
This ，it's jpeg data. But this one via UART ? not spi. I think. :D

sakabin, 2019-04-11:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals
Use spi send fun instead uart

Herwey, 2019-04-11:
Hi:
I printed the jpeg fb, it includes int numbers. What the meaning of the these numbers? can i read the rgb info from the numbers? :P

closed 2019-04-12 by Herwey