m5stack/M5Stack-Camera #21: ESP32-CAM without psram

thbeh, 2019-12-04:
Hi, what happen to this https://github.com/m5stack/m5stack-cam-psram/tree/NoPsram? What code should I use to flash non psram m5stack esp32-cam? Thanks in advance.

EeeeBin, 2019-12-04:
https://github.com/m5stack/m5stack-cam-psram/tree/master/wifi/wifi_ap/firmware/ESP32-Camera
Run **_make menuconfig_** and select _**ESP32-Camera**_

thbeh, 2019-12-04:
I did that but I get this when I make monitor -
(python2.7) thbeh@beary-x99a:~/Downloads/esp-idf-v3.3/examples/get-started/m5stack-cam-psram/wifi/wifi_sta$ make monitor
Toolchain path: /home/thbeh/Downloads/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /home/thbeh/Downloads/esp-idf-v3.3/requirements.txt are satisfied.
MONITOR
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6352
ho 0 tail 12 room 4
load:0x40078000,len:11624
ho 0 tail 12 room 4
load:0x40080400,len:6648
entry 0x40080764
I (32) boot: ESP-IDF v3.3 2nd stage bootloader
I (32) boot: compile time 22:30:29
I (32) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x26c14 (158740) map
I (151) esp_image: segment 1: paddr=0x00036c3c vaddr=0x3ffb0000 size=0x038a0 ( 14496) load
I (157) esp_image: segment 2: paddr=0x0003a4e4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/thbeh/Downloads/esp-idf-v3.3/components/freertos/xtensa_vectors.S:1779

I (158) esp_image: segment 3: paddr=0x0003a8ec vaddr=0x40080400 size=0x05724 ( 22308) load
I (176) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x7ed70 (519536) map
0x400d0018: _stext at ??:?

I (358) esp_image: segment 5: paddr=0x000bed90 vaddr=0x40085b24 size=0x11cdc ( 72924) load
0x40085b24: esp_cpu_unstall at /home/thbeh/Downloads/esp-idf-v3.3/components/soc/esp32/cpu_util.c:40

I (402) boot: Loaded app from partition at offset 0x10000
I (402) boot: Disabling RNG early entropy source...
I (403) psram: This chip is ESP32-D0WD
E (409) spiram: SPI RAM enabled but initialization failed. Bailing out.
I (414) cpu_start: Failed to init external RAM; continuing without it.
I (422) cpu_start: Pro cpu up.
I (425) cpu_start: Application information:
I (430) cpu_start: Project name:     camera
I (435) cpu_start: App version:      1c0504d-dirty
I (441) cpu_start: Compile time:     Dec  3 2019 22:30:35
I (447) cpu_start: ELF file SHA256:  4cf0ff92869b7c0e...
I (453) cpu_start: ESP-IDF:          v3.3
I (457) cpu_start: Starting app cpu, entry point is 0x40081520
0x40081520: call_start_cpu1 at /home/thbeh/Downloads/esp-idf-v3.3/components/esp32/cpu_start.c:270

I (0) cpu_start: App cpu up.
I (468) heap_init: Initializing. RAM available for dynamic allocation:
I (475) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (481) heap_init: At 3FFBB878 len 00024788 (145 KiB): DRAM
I (487) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (493) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (500) heap_init: At 40097800 len 00008800 (34 KiB): IRAM
I (506) cpu_start: Pro cpu start user code
I (189) cpu_start: Chip Revision: 1
W (189) cpu_start: Chip revision is higher than the one configured in menuconfig. Suggest to upgrade it.
I (193) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (203) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (243) sccb: pin_sda 25 pin_scl 23

I (243) gpio: GPIO[0]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (273) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (323) camera: Detected camera not supported.
E (323) camera: Camera probe failed with error 0x20004
E (323) camera: Camera Init Failed

EeeeBin, 2019-12-04:
After testing, it is found that the firmware is running normally
Have you used this firmware: 
https://github.com/m5stack/m5stack-cam-psram/tree/master/wifi/wifi_ap/firmware/ESP32-Camera
If YES, pls confirm hardware contact is normal

thbeh, 2019-12-04:
I can't use that because I am using wifi_sta (i need to put my ssid and password) but if I use firmware for wifi_ap it runs fine.

EeeeBin, 2019-12-04:
**_make menuconfig_** --> **_Camera model_** --> **_ESP32-Camera_**
**_make menuconfig_** --> **_Component config_** --> **_Camera..._** --> **_Use hardware I2C1 for SCCB = n,  OV2640 Support  = y_**
And test
 If failed, can copy sta project main.c to  ap project, and Compile ap project

thbeh, 2019-12-04:
I did as per your instructions and it work but the when I point to the cam using chrome, it constantly flicker. Before testing your instructions, i tried with **make support for external, SPI-connected RAM = n** and it worked too but the screen flicker too.

Somehow using AP mode totally does not flicker regardless what framesize if set. Any thoughts?

EeeeBin, 2019-12-05:
The STA mode depends on the LAN environment, so flicker is normal

closed 2019-12-05 by EeeeBin