m5stack/M5Stack-Camera #23: esp-32 cam, no psram, compiling error

CostInCode, 2019-12-13:
Using eps-idf latest version (4.1-dev), I'm trying to build the firmware for esp-32 cam without psram but I'm get a compiler error.
Configuration options
cd m5stack-cam-psram/wifi/wifi_ap/
make menuconfig --> camera model --> esp32camera
make menuconfig --> component config --> camera... --> Use hardware I2C1 for SCCB = n, OV2640 Support = y
make menuconfig --> component config --> esp32 specific --> support for external spi ram = y --> spi ram config --> Ignore psram when not found = y
All the other options as default.

**make gives the following error**: 

> /esp-idf/m5stack-cam-psram/wifi/wifi_ap/main/main.c:232:17: error: incompatible types when assigning to type 'ip4_addr_t' {aka 'struct ip4_addr'} from type 'esp_ip4_addr_t' {aka 'struct esp_ip4_addr'}
       s_ip_addr = event->event_info.got_ip.ip_info.ip;

CostInCode, 2019-12-16:
That was my mistake, with `make flash` and running flash script it worked.

closed 2019-12-16 by CostInCode

FunnyDevs, 2019-12-20:
I have same problem and i was not able to resolve.

haixuanTao, 2020-02-22:
Solved it by changing:
```
static ip4_addr_t s_ip_addr;
```
to
```
static esp_ip4_addr_t s_ip_addr;
```
in file `m5stack-cam-psram/wifi/wifi_ap/main/main.c:232:17`