m5stack/M5Stack-Camera #24: undefined reference to `led_brightness'

vaderj, 2020-02-09:
ESP-IDF: 3.2.3
Windows 10
I am attempting to flash the WIFI_AP example to my M5Stack ESP32 Cam but the build keeps failing with a message about the led_brightness.

The only change I made to the main.c of the WIFI_AP example is I changed the SSID name, just to see if it would build.

The "idf.py build" output is as follows:

c:\SRC\m5stack-cam-psram\wifi\wifi_ap>idf.py build
Note: You are using Python 3.7.4. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from c:\esp\esp-idf\requirements.txt are satisfied.
Running ninja in directory c:\SRC\m5stack-cam-psram\wifi\wifi_ap\build
Executing "ninja all"...
[494/751] Performing configure step for 'bootloader'
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.24.0.windows.2")
-- Component names: soc spi_flash micro-ecc bootloader_support esptool_py partition_table main esp32 log bootloader
-- Component paths: C:/esp/esp-idf/components/soc;C:/esp/esp-idf/components/spi_flash;C:/esp/esp-idf/components/micro-ecc;C:/esp/esp-idf/components/bootloader_support;C:/esp/esp-idf/components/esptool_py;C:/esp/esp-idf/components/partition_table;C:/esp/esp-idf/components/bootloader/subproject/main;C:/esp/esp-idf/components/esp32;C:/esp/esp-idf/components/log;C:/esp/esp-idf/components/bootloader
-- The ASM compiler identification is GNU
-- Found assembler: C:/esp/tools/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- Check for working C compiler: C:/esp/tools/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/esp/tools/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/esp/tools/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/esp/tools/tools/xtensa-esp32-elf/1.22.0-80-g6c4433a5-5.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- ccache will be used for faster builds
-- Component libraries: soc;spi_flash;micro-ecc;bootloader_support;main;esp32;log
-- Adding linker script C:/esp/esp-idf/components/bootloader/subproject/main/esp32.bootloader.ld
-- Adding linker script C:/esp/esp-idf/components/bootloader/subproject/main/esp32.bootloader.rom.ld
-- Adding linker script C:/esp/esp-idf/components/esp32/ld/esp32.rom.ld
-- Adding linker script C:/esp/esp-idf/components/esp32/ld/esp32.rom.spiram_incompatible_fns.ld
-- Adding linker script C:/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Configuring done
-- Generating done
-- Build files have been written to: C:/SRC/m5stack-cam-psram/wifi/wifi_ap/build/bootloader
[541/751] Performing build step for 'bootloader'
[1/43] Building C object soc/CMakeFiles/soc.dir/esp32/gpio_periph.c.obj
[2/43] Building C object soc/CMakeFiles/soc.dir/esp32/rtc_init.c.obj
[3/43] Building C object soc/CMakeFiles/soc.dir/esp32/cpu_util.c.obj
[4/43] Building C object soc/CMakeFiles/soc.dir/esp32/rtc_clk_init.c.obj
[5/43] Building C object soc/CMakeFiles/soc.dir/esp32/rtc_clk.c.obj
[6/43] Building C object soc/CMakeFiles/soc.dir/esp32/rtc_periph.c.obj
[7/43] Building C object soc/CMakeFiles/soc.dir/esp32/rtc_sleep.c.obj
[8/43] Building C object soc/CMakeFiles/soc.dir/esp32/sdio_slave_periph.c.obj
[9/43] Building C object soc/CMakeFiles/soc.dir/esp32/rtc_pm.c.obj
[10/43] Building C object soc/CMakeFiles/soc.dir/esp32/rtc_wdt.c.obj
[11/43] Generating dummy_main_src.c
[12/43] Building C object soc/CMakeFiles/soc.dir/esp32/sdmmc_periph.c.obj
[13/43] Building C object soc/CMakeFiles/soc.dir/esp32/rtc_time.c.obj
[14/43] Building C object soc/CMakeFiles/soc.dir/esp32/soc_memory_layout.c.obj
[15/43] Building C object soc/CMakeFiles/soc.dir/esp32/spi_periph.c.obj
[16/43] Building C object CMakeFiles/bootloader.elf.dir/dummy_main_src.c.obj
[17/43] Building C object spi_flash/CMakeFiles/spi_flash.dir/spi_flash_rom_patch.c.obj
[18/43] Linking C static library spi_flash\libspi_flash.a
[19/43] Building C object micro-ecc/CMakeFiles/micro-ecc.dir/micro-ecc/uECC.c.obj
[20/43] Building C object soc/CMakeFiles/soc.dir/src/memory_layout_utils.c.obj
[21/43] Building C object bootloader_support/CMakeFiles/bootloader_support.dir/src/bootloader_clock.c.obj
[22/43] Linking C static library micro-ecc\libmicro-ecc.a
[23/43] Linking C static library soc\libsoc.a
[24/43] Building C object bootloader_support/CMakeFiles/bootloader_support.dir/src/bootloader_common.c.obj
[25/43] Building C object bootloader_support/CMakeFiles/bootloader_support.dir/src/efuse.c.obj
[... 17 lines cut ...]
[43/43] Generating bootloader.bin
esptool.py v2.6
[748/751] Building C object main/CMakeFiles/main.dir/main.c.obj
In file included from ../main/main.c:17:0:
../main/config.h:31:0: warning: "CAM_PIN_D0" redefined
     #define CAM_PIN_D0      17
 ^
../main/config.h:19:0: note: this is the location of the previous definition
 #define CAM_PIN_D0      32
 ^
../main/config.h:36:0: warning: "CAMERA_LED_GPIO" redefined
     #define CAMERA_LED_GPIO 16
 ^
../main/config.h:25:0: note: this is the location of the previous definition
 #define CAMERA_LED_GPIO 14
 ^
[750/751] Linking CXX executable camera.elf
FAILED: camera.elf
cmd.exe /C "cd . && C:\esp\tools\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe    -nostdlib CMakeFiles/camera.elf.dir/dummy_main_src.c.obj  -o camera.elf  -Wl,--gc-sections -Wl,--cref -Wl,--Map=camera.map -Wl,--start-group soc/libsoc.a log/liblog.a heap/libheap.a xtensa-debug-module/libxtensa-debug-module.a app_trace/libapp_trace.a freertos/libfreertos.a vfs/libvfs.a newlib/libnewlib.a esp_ringbuf/libesp_ringbuf.a driver/libdriver.a esp_event/libesp_event.a ethernet/libethernet.a lwip/liblwip.a tcpip_adapter/libtcpip_adapter.a app_update/libapp_update.a spi_flash/libspi_flash.a mbedtls/libmbedtls.a micro-ecc/libmicro-ecc.a bootloader_support/libbootloader_support.a nvs_flash/libnvs_flash.a pthread/libpthread.a smartconfig_ack/libsmartconfig_ack.a wpa_supplicant/libwpa_supplicant.a esp32/libesp32.a cxx/libcxx.a esp32-camera/libesp32-camera.a asio/libasio.a jsmn/libjsmn.a coap/libcoap.a console/libconsole.a nghttp/libnghttp.a esp-tls/libesp-tls.a esp_adc_cal/libesp_adc_cal.a tcp_transport/libtcp_transport.a esp_http_client/libesp_http_client.a esp_http_server/libesp_http_server.a esp_https_ota/libesp_https_ota.a expat/libexpat.a wear_levelling/libwear_levelling.a sdmmc/libsdmmc.a fatfs/libfatfs.a freemodbus/libfreemodbus.a json/libjson.a libsodium/liblibsodium.a mdns/libmdns.a mqtt/libmqtt.a openssl/libopenssl.a protobuf-c/libprotobuf-c.a protocomm/libprotocomm.a spiffs/libspiffs.a ulp/libulp.a wifi_provisioning/libwifi_provisioning.a main/libmain.a -lgcov -Wl,--undefined=uxTopUsedPriority -L C:/esp/esp-idf/components/newlib/lib -lc-psram-workaround -lm-psram-workaround -L C:/esp/esp-idf/components/esp32/lib -L C:/esp/esp-idf/components/esp32/ld/wifi_iram_opt -lcoexist -lcore -lespnow -lmesh -lnet80211 -lphy -lpp -lrtc -lsmartconfig -lwpa2 -lwpa -lwps -L C:/SRC/m5stack-cam-psram/wifi/wifi_ap/build/esp32 -T esp32_out.ld -L C:/esp/esp-idf/components/esp32/ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld C:/esp/esp-idf/components/esp32/libhal.a -lgcc -u call_user_start_cpu0 -u ld_include_panic_highint_hdl -lstdc++ -u __cxa_guard_dummy -u __cxx_fatal_exception && cd ."
main/libmain.a(main.c.obj):(.literal.app_main+0x18): undefined reference to `led_brightness'
main/libmain.a(main.c.obj): In function `app_main':
c:\SRC\m5stack-cam-psram\wifi\wifi_ap\build/../main/main.c:82: undefined reference to `led_brightness'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

vaderj, 2020-02-15:
I've also tried 3.2.2 with the same result

EeeeBin, 2020-02-15:
pls try use make instead cmake，run  `make`

vaderj, 2020-02-16:
That ended up doing it - using IDFv3.3.1 and the msys32 tools I was able to compile and flash!

closed 2020-02-27 by vaderj