m5stack/M5Stack-Camera #25: issue with "make"  /uart/components/esp32-camera/driver/twi.c', needed by `driver/twi.o'.

d3-liubomyr-kozak, 2020-02-22:
module: ESP32CAM
sdkconfig: default
tested code: https://github.com/m5stack/m5stack-cam-psram/tree/master/uart

ESP-IDF v3.2.3-167-g3b7777e22-dirty 2nd stage bootloader
when try make flash getting error make[1]: *** No rule to make target `/......./m5stack-cam-psram/uart/components/esp32-camera/driver/twi.c', needed by `driver/twi.o'.  Stop.
make: *** [component-esp32-camera-build] Error 2

chidea, 2020-06-09:
according to [this esp32-homekit-camera issue](https://github.com/maximkulkin/esp32-homekit-camera/issues/57), go to `menuconfig > Component config > driver configurations > RTCI0 configuration` and enable it, save and build.

liubomyr-kozak, 2020-06-19:
work for me, thanks

closed 2020-06-19 by liubomyr-kozak