m5stack/M5Stack-Camera #28: 835/836 FAILED: camera.elf

Tchill73, 2020-04-15:
Hello,

I have a camera M5stack F and i want to flah with wifi_sta (Windows 10 & ESP-IDF 4.0)  but i can't do it, Can you help me please, I'm a Young user.

`C:\ESP-IDF\m5\wifi\wifi_sta>idf.py set-target esp32
Checking Python dependencies...
Python requirements from C:\ESP-IDF\requirements.txt are satisfied.
Usage: idf.py [OPTIONS] COMMAND1 [ARGS]... [COMMAND2 [ARGS]...]...
Try 'idf.py --help' for help.

Error: No such command 'set-target'.

C:\ESP-IDF\m5\wifi\wifi_sta>idf.py flash
Checking Python dependencies...
Python requirements from C:\ESP-IDF\requirements.txt are satisfied.
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running cmake in directory c:\esp-idf\m5\wifi\wifi_sta\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized -DCCACHE_ENABLE=0 c:\esp-idf\m5\wifi\wifi_sta"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.21.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project version: v4.0-dirty
-- Building ESP-IDF components for target esp32
-- Found PythonInterp: C:/.espressif/python_env/idf4.0_py3.8_env/Scripts/python.exe (found version "3.8.2")
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- Adding linker script C:/ESP-IDF/m5/wifi/wifi_sta/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/ESP-IDF/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/ESP-IDF/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/ESP-IDF/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/ESP-IDF/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/ESP-IDF/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/ESP-IDF/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/ESP-IDF/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt coap console cxx driver efuse esp-tls esp32 esp32-camera esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/ESP-IDF/components/app_trace C:/ESP-IDF/components/app_update C:/ESP-IDF/components/asio C:/ESP-IDF/components/bootloader C:/ESP-IDF/components/bootloader_support C:/ESP-IDF/components/bt C:/ESP-IDF/components/coap C:/ESP-IDF/components/console C:/ESP-IDF/components/cxx C:/ESP-IDF/components/driver C:/ESP-IDF/components/efuse C:/ESP-IDF/components/esp-tls C:/ESP-IDF/components/esp32 C:/ESP-IDF/m5/wifi/wifi_sta/components/esp32-camera C:/ESP-IDF/components/esp_adc_cal C:/ESP-IDF/components/esp_common C:/ESP-IDF/components/esp_eth C:/ESP-IDF/components/esp_event C:/ESP-IDF/components/esp_gdbstub C:/ESP-IDF/components/esp_http_client C:/ESP-IDF/components/esp_http_server C:/ESP-IDF/components/esp_https_ota C:/ESP-IDF/components/esp_https_server C:/ESP-IDF/components/esp_local_ctrl C:/ESP-IDF/components/esp_ringbuf C:/ESP-IDF/components/esp_rom C:/ESP-IDF/components/esp_websocket_client C:/ESP-IDF/components/esp_wifi C:/ESP-IDF/components/espcoredump C:/ESP-IDF/components/esptool_py C:/ESP-IDF/components/expat C:/ESP-IDF/components/fatfs C:/ESP-IDF/components/freemodbus C:/ESP-IDF/components/freertos C:/ESP-IDF/components/heap C:/ESP-IDF/components/idf_test C:/ESP-IDF/components/jsmn C:/ESP-IDF/components/json C:/ESP-IDF/components/libsodium C:/ESP-IDF/components/log C:/ESP-IDF/components/lwip C:/ESP-IDF/m5/wifi/wifi_sta/main C:/ESP-IDF/components/mbedtls C:/ESP-IDF/components/mdns C:/ESP-IDF/components/mqtt C:/ESP-IDF/components/newlib C:/ESP-IDF/components/nghttp C:/ESP-IDF/components/nvs_flash C:/ESP-IDF/components/openssl C:/ESP-IDF/components/partition_table C:/ESP-IDF/components/protobuf-c C:/ESP-IDF/components/protocomm C:/ESP-IDF/components/pthread C:/ESP-IDF/components/sdmmc C:/ESP-IDF/components/soc C:/ESP-IDF/components/spi_flash C:/ESP-IDF/components/spiffs C:/ESP-IDF/components/tcp_transport C:/ESP-IDF/components/tcpip_adapter C:/ESP-IDF/components/ulp C:/ESP-IDF/components/unity C:/ESP-IDF/components/vfs C:/ESP-IDF/components/wear_levelling C:/ESP-IDF/components/wifi_provisioning C:/ESP-IDF/components/wpa_supplicant C:/ESP-IDF/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/ESP-IDF/m5/wifi/wifi_sta/build
Running ninja in directory c:\esp-idf\m5\wifi\wifi_sta\build
Executing "ninja all"...
[655/836] Performing configure step for 'bootloader'
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.21.0.windows.1")
-- mconf-idf version mconf-v4.6.0.0-idf-20190628-win32
-- The C compiler identification is GNU 8.2.0
-- The CXX compiler identification is GNU 8.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
[... 101 lines cut ...]
  ^~~~~~~
In file included from ../components/esp32-camera/driver/private_include/camera_common.h:6,
                 from ../components/esp32-camera/driver/camera.c:35:
c:\esp-idf\components\esp32\include\rom\lldesc.h:1:2: warning: #warning rom/lldesc.h is deprecated, please use esp32/rom/lldesc.h instead [-Wcpp]
 #warning rom/lldesc.h is deprecated, please use esp32/rom/lldesc.h instead
  ^~~~~~~
[827/836] Building C object esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/conversions/to_bmp.c.obj
In file included from ../components/esp32-camera/conversions/to_bmp.c:17:
c:\esp-idf\components\esp32\include\esp_spiram.h:1:2: warning: #warning esp_spiram.h has been replaced by esp32/spiram.h, please include esp32/spiram.h instead [-Wcpp]
 #warning esp_spiram.h has been replaced by esp32/spiram.h, please include esp32/spiram.h instead
  ^~~~~~~
[829/836] Building CXX object esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/conversions/to_jpg.cpp.obj
In file included from ../components/esp32-camera/conversions/to_jpg.cpp:16:
c:\esp-idf\components\esp32\include\esp_spiram.h:1:2: warning: #warning esp_spiram.h has been replaced by esp32/spiram.h, please include esp32/spiram.h instead [-Wcpp]
 #warning esp_spiram.h has been replaced by esp32/spiram.h, please include esp32/spiram.h instead
  ^~~~~~~
[835/836] Linking CXX executable camera.elf
FAILED: camera.elf
cmd.exe /C "cd . && C:\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\camera.elf.rsp  -o camera.elf  && cd ."
[... cut ...] (main.c.obj):(.literal.app_main+0x18): undefined reference to `led_brightness'
c:/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(main.c.obj): in function `app_main':
c:\esp-idf\m5\wifi\wifi_sta\build/../main/main.c:89: undefined reference to `led_brightness'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

EeeeBin, 2020-04-15:
Sorry, this lib not support idf 4.0 yet
Try use 3.3-rc, make (not cmake)

Tchill73, 2020-04-18:
Hello, thank your for reply.

 I try with ESP-IDF (Windows 10)
5stack-cam-psram-master\m5stack-cam-psram-master\wifi\wifi_sta\firmware\Camera F
I am sure there is not much missing

The error is :
'FAILED:` esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj
ccache C:.espressif\tools\xtensa-esp32-elf\esp-2020r1-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -I../../../../components/newlib/platform_include -I../../../../components/freertos/include -I../../../../components/freertos/xtensa/include -I../../../../components/heap/include -I../../../../components/log/include -I../../../../components/lwip/include/apps -I../../../../components/lwip/include/apps/sntp -I../../../../components/lwip/lwip/src/include -I../../../../components/lwip/port/esp32/include -I../../../../components/lwip/port/esp32/include/arch -I../../../../components/soc/src/esp32/. -I../../../../components/soc/src/esp32/include -I../../../../components/soc/include -I../../../../components/esp_rom/include -I../../../../components/esp_common/include -I../../../../components/esp_system/include -I../../../../components/xtensa/include -I../../../../components/xtensa/esp32/include -I../../../../components/esp32/include -I../../../../components/driver/include -I../../../../components/driver/esp32/include -I../../../../components/esp_ringbuf/include -I../../../../components/efuse/include -I../../../../components/efuse/esp32/include -I../../../../components/espcoredump/include -I../../../../components/esp_timer/include -I../../../../components/soc/soc/esp32/. -I../../../../components/soc/soc/esp32/include -I../../../../components/soc/soc/esp32/../include -I../../../../components/vfs/include -I../../../../components/esp_wifi/include -I../../../../components/esp_wifi/esp32/include -I../../../../components/esp_event/include -I../../../../components/esp_netif/include -I../../../../components/esp_eth/include -I../../../../components/tcpip_adapter/include -I../../../../components/app_trace/include -I../../../../components/mbedtls/port/include -I../../../../components/mbedtls/mbedtls/include -I../../../../components/mbedtls/esp_crt_bundle/include -I../../../../components/bootloader_support/include -I../../../../components/app_update/include -I../../../../components/spi_flash/include -I../../../../components/wpa_supplicant/include -I../../../../components/wpa_supplicant/port/include -I../../../../components/wpa_supplicant/include/esp_supplicant -I../../../../components/nvs_flash/include -I../../../../components/pthread/include -I../../../../components/perfmon/include -I../../../../components/asio/asio/asio/include -I../../../../components/asio/port/include -I../../../../components/cbor/port/include -I../../../../components/coap/port/include -I../../../../components/coap/port/include/coap -I../../../../components/coap/libcoap/include -I../../../../components/coap/libcoap/include/coap2 -I../../../../components/console -I../../../../components/nghttp/port/include -I../../../../components/nghttp/nghttp2/lib/includes -I../../../../components/esp-tls -I../../../../components/esp_adc_cal/include -I../../../../components/esp_gdbstub/include -I../../../../components/tcp_transport/include -I../../../../components/esp_http_client/include -I../../../../components/esp_http_server/include -I../../../../components/esp_https_ota/include -I../../../../components/protobuf-c/protobuf-c -I../../../../components/protocomm/include/common -I../../../../components/protocomm/include/security -I../../../../components/protocomm/include/transports -I../../../../components/mdns/include -I../../../../components/esp_local_ctrl/include -I../../../../components/sdmmc/include -I../../../../components/esp_serial_slave_link/include -I../../../../components/esp_websocket_client/include -I../../../../components/expat/expat/expat/lib -I../../../../components/expat/port/include -I../../../../components/wear_levelling/include -I../../../../components/fatfs/diskio -I../../../../components/fatfs/vfs -I../../../../components/fatfs/src -I../../../../components/freemodbus/common/include -I../../../../components/idf_test/include -I../../../../components/idf_test/include/esp32 -I../../../../components/jsmn/include -I../../../../components/json/cJSON -I../../../../components/libsodium/libsodium/src/libsodium/include -I../../../../components/libsodium/port_include -I../../../../components/mqtt/esp-mqtt/include -I../../../../components/openssl/include -I../../../../components/spiffs/include -I../../../../components/ulp/include -I../../../../components/unity/include -I../../../../components/unity/unity/src -I../../../../components/wifi_provisioning/include -I../components/esp32-camera/driver/include -I../components/esp32-camera/conversions/include -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER="v4.2-dev-1126-gd85d3d969" -DESP_PLATFORM -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj -MF esp-idf\main\CMakeFiles__idf_main.dir\main.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/main.c.obj -c ../main/main.c
In file included from ../main/main.c:15:
[... cut ...]
[927/936] Building CXX object esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/conversions/to_jpg.cpp.obj
In file included from ../components/esp32-camera/conversions/to_jpg.cpp:16:
../../../../components/esp32/include/esp_spiram.h:1:2: warning: #warning esp_spiram.h has been replaced by esp32/spiram.h, please include esp32/spiram.h instead [-Wcpp]
#warning esp_spiram.h has been replaced by esp32/spiram.h, please include esp32/spiram.h instead'
ninja: build stopped: subcommand failed.
ninja failed with exit code '

Tchill73, 2020-04-19:
In the begining with esp-idf v4.2
And with esp-idf 3.3 realease
He can't finish "build"

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool xtensa-esp32-elf found in PATH: esp-2020r1-8.2.0.
Not using an unsupported version of tool cmake found in PATH: 3.16.4.
Not using an unsupported version of tool openocd-esp32 found in PATH: v0.10.0-esp32-20200406.
Not using an unsupported version of tool ninja found in PATH: 1.10.0.
C:\Users\oxyrace.espressif\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin
C:\Users\oxyrace.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin
C:\Users\oxyrace.espressif\tools\cmake\3.13.4\bin
C:\Users\oxyrace.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin
C:\Users\oxyrace.espressif\tools\mconf\v4.6.0.0-idf-20190628
C:\Users\oxyrace.espressif\tools\ninja\1.9.0
C:\Users\oxyrace.espressif\tools\idf-exe\1.0.1
C:\Users\oxyrace.espressif\tools\ccache\3.7
C:\Users\oxyrace.espressif\python_env\idf3.3_py3.8_env\Scripts
[836/837] Linking CXX executable camera.elf
FAILED: camera.elf
cmd.exe /C "cd . && C:\Users\oxyrace.espressif\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -nostdlib @CMakeFiles\camera.elf.rsp -o camera.elf && cd ."
esp-idf/main/libmain.a(main.c.obj):(.literal.app_main+0x18): undefined reference to led_brightness' esp-idf/main/libmain.a(main.c.obj): In functionapp_main':
D:\ESP32\esp-idf\examples\m5stack\wifi\wifi_sta\build/../main/main.c:89: undefined reference to `led_brightness'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1