m5stack/M5Stack-Camera #31: JPEG header not correct

davidkingzyb, 2020-06-23:
My m5stack camera F not work, when check JPEG header.

`camera.c` file line 724
```
       //check for correct JPEG header
        if(s_state->sensor.pixformat == PIXFORMAT_JPEG) {
            uint32_t sig = *((uint32_t *)s_state->fb->buf) & 0xFFFFFF;
            if(sig != 0xffd8ff) {//sig always 0xffd9ff
                s_state->fb->bad = 1;
                return;
            }
        }
```
Is it a hardware problem?
If not how to fix it?

EeeeBin, 2020-06-24:
Maybe software problem, 
Try test https://docs.m5stack.com/#/en/unit/m5camera_f?id=easyloader, 
if it works well, pls describe the software used
if works wrong, it`s hardware problem

davidkingzyb, 2020-06-24:
thx. 
But it is not works. 
Looks like my M5 broken.

closed 2020-06-24 by davidkingzyb