m5stack/M5Stack-Camera #33: incompatible types when assigning to type 'ip4_addr_t'

dcmcshan, 2020-06-27:
M5Stack-Camera/qr/main/main.c:274:17: error: incompatible types when assigning to type 'ip4_addr_t' {aka 'struct ip4_addr'} from type 'esp_ip4_addr_t' {aka 'struct esp_ip4_addr'}
       s_ip_addr = event->event_info.got_ip.ip_info.ip;
                 ^

EeeeBin, 2020-06-28:
QR example now only support idf3.3

kenken64, 2020-11-07:
C:/Users/kenke/Desktop/esp-idf/components/lwip/lwip/src/include/lwip/ip4_addr.h:207:7: note: expected 'const ip4_addr_t *' {aka 'const struct ip4_addr *'} but argument is of type 'esp_ip4_addr_t *' {aka 'struct esp_ip4_addr *'}
 char *ip4addr_ntoa(const ip4_addr_t *addr);
       ^~~~~~~~~~~~
../main/main.c:239:17: error: incompatible types when assigning to type 'ip4_addr_t' {aka 'struct ip4_addr'} from type 'esp_ip4_addr_t' {aka 'struct esp_ip4_addr'}
       s_ip_addr = event->event_info.got_ip.ip_info.ip;
                 ^
../main/main.c: In function 'wifi_init_softap':
../main/main.c:266:3: warning: 'tcpip_adapter_init' is deprecated [-Wdeprecated-declarations]
   tcpip_adapter_init();
   ^~~~~~~~~~~~~~~~~~
In file included from C:/Users/kenke/Desktop/esp-idf/components/esp_netif/include/esp_netif.h:35,
                 from C:/Users/kenke/Desktop/esp-idf/components/esp_event/include/esp_event_legacy.h:22,
                 from C:/Users/kenke/Desktop/esp-idf/components/esp_event/include/esp_event.h:26,
                 from C:/Users/kenke/Desktop/esp-idf/components/esp_wifi/include/esp_wifi.h:64,
                 from ../main/main.c:11:
C:/Users/kenke/Desktop/esp-idf/components/tcpip_adapter/include/tcpip_adapter.h:34:6: note: declared here
 void tcpip_adapter_init(void)  __attribute__ ((deprecated));
      ^~~~~~~~~~~~~~~~~~
../main/main.c:267:3: warning: 'esp_event_loop_init' is deprecated [-Wdeprecated-declarations]
   ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));
   ^~~~~~~~~~~~~~~
In file included from C:/Users/kenke/Desktop/esp-idf/components/esp_event/include/esp_event.h:26,
                 from C:/Users/kenke/Desktop/esp-idf/components/esp_wifi/include/esp_wifi.h:64,
                 from ../main/main.c:11:
C:/Users/kenke/Desktop/esp-idf/components/esp_event/include/esp_event_legacy.h:227:11: note: declared here
 esp_err_t esp_event_loop_init(system_event_cb_t cb, void *ctx) __attribute__ ((deprecated));
           ^~~~~~~~~~~~~~~~~~~
[709/781] Building C object esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/sensors/ov5640.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

ZRock0, 2022-03-12:
and how to fix?

ZRock0, 2022-03-12:
just delete