m5stack/M5Stack-Camera #34: error: 'GPIO_PIN_COUNT' <> 'SOC_GPIO_PIN_COUNT'

ChristophNoack, 2020-11-07:
Hi,

I try to flash my new M5Stack ESP32 cam.

While making my project I get the following error:
`error: 'GPIO_PIN_COUNT' undeclared here (not in a function); did you mean 'SOC_GPIO_PIN_COUNT'?`
I checked the rtc_gpio_desc point in the menuconfig.

Could you tell me what I am doing wrong?

kenken64, 2020-11-07:
./components/esp32-camera/driver/twi.c: In function 'pinMode':
../components/esp32-camera/driver/twi.c:61:24: error: 'rtc_gpio_desc' undeclared (first use in this function); did you mean 'rtc_io_desc'?
     uint32_t rtc_reg = rtc_gpio_desc[pin].reg;
                        ^~~~~~~~~~~~~
                        rtc_io_desc
../components/esp32-camera/driver/twi.c:61:24: note: each undeclared identifier is reported only once for each function it appears in
[1110/1179] Building C object esp-idf/esp32-camera/CMakeFiles/__idf_esp32-camera.dir/sensors/ov5640.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

ChristophNoack, 2020-11-07:
Excuse me, I do not understand the answer. 
When I uncheck the rtc_gpio_desc point then I get a similar error like 'rtc_gpio_desc' undeclared (first use in this function); did you mean 'rtc_io_desc'?

But this doesn't help me with my problem...

kenken64, 2020-11-08:
chris,

i am facing the same issue thats why i post here

kenken64, 2020-11-08:
which idf sdk you are using ? i am using idf 4.1 and python 2.7

kenken64, 2020-11-08:
hey let me try this as well , i think is related to this issue 

https://github.com/m5stack/M5Stack-Camera/issues/29

ChristophNoack, 2020-11-08:
Ah ok. :)
I use idf4.3 and python 2.7

ChristophNoack, 2020-11-08:
Could plz so help us?

kenken64, 2020-11-09:
follow the idf menuconfig for #29

ChristophNoack, 2020-11-09:
I do but it doesn't work... :-(

mws-rmain, 2020-11-10:
Please post the all of the lines from the build log showing the error.
There should be several lines before & possibly after:
`error: 'GPIO_PIN_COUNT' undeclared here (not in a function); did you mean 'SOC_GPIO_PIN_COUNT'?`
that provide relevant details.

EeeeBin, 2020-11-11:
HI,  Due to the incompatible update of ESP-IDF, the current code only supports idf4.0 version now
I am also looking for a code that can be version compatible with multiple IDF versions

ChristophNoack, 2020-11-23:
That's it. I had to chose ESP-IDF4.0!
Thanks!